BCLab-UNM/SwarmBaseCode-ROS #26: Rover startup scripts

gmfricke, 2017-09-13:
Please create scripts that allow rovers to be loaded with the code the user selects and the ROS nodes started without the user having to manually ssh to the rovers. This script should handle multiple rovers.

mike-matera, 2017-09-15:
I just got this working, but the changes are extensive. I can send you a pull request if you like. It'd help me to avoid upstream conflicts.

jschlind411, 2017-09-18:
StartupScript branch was temporarily made so others can checkout code.

I will clean it up a little bit before I have finalized it.  Look for deploy.sh under misc folder

File accepts 3 commands (-G, -F, and -R)

currently debugging 2 blockers:

1.) After sending over repository using scp rover sometimes needs to be rebooted before the rover GUI will update with info (navsat, odom, map, and link quality).  A reboot fixes the problem but unsure why.

2.) When running deploy.sh from different machines I have gotten an EOF error when deploying the rover_onboard_node_launch.sh... still diagnosing this problem

3.) Been having an issue with pinging rovers before running scripts.  
     - The original deploy_kinetic script I was given the ping worked.  However, been having issues with it not running the appropriate code if pinged?  Current priority.

Those 3 blockers are the primary reason for delay.  

Once blockers are solved, will clean and request approval!

wfvining, 2017-10-30:
This was done some time ago (pr #48)

closed 2017-10-30 by wfvining