BCLab-UNM/SwarmBaseCode-ROS #29: Different decelerations for moving forward and backward when controlling manually

jprod123, 2017-09-14:
When driving the swarmie manually via joystick moving backward has a sharp deceleration making it stop exactly where intended. Moving forward however takes ~2 seconds to fully decelerate causing overshoots when driving forward but also causes over-steering as well. When turning, the wheel rotating forward take longer to decelerate causing it to over steer.

mike-matera, 2017-09-15:
This happens because abridge should be sending the stop ("s") command to Arduino when the requested velocity is zero. Try changing code in abridge.cpp to look like this:

```c++
  int leftInt = left;
  int rightInt = right;

  if (leftInt == 0 && rightInt == 0) {    
    sprintf(moveCmd, "s\n");
  }
  else {
    sprintf(moveCmd, "v,%d,%d\n", leftInt, rightInt); //format data for arduino into c string
  }
  usb.sendData(moveCmd);                      //send movement command to arduino over usb
  memset(&moveCmd, '\0', sizeof (moveCmd));   //clear the movement command string
```

jprod123, 2017-09-15:
This fixes the issue. Thanks!
Just some side observations: Before the fix driving backwards had a very sharp stop. This is not the case anymore. But the forward and backward deceleration are the same now fixing the skid-steering and the Swarmie is perfectly drivable.

closed 2017-09-18 by jprod123