BCLab-UNM/SwarmBaseCode-ROS #58: Rover overshoots waypoints

JCHJones, 2017-10-09:
Rovers will occasionally shoot past their waypoints. This overshoot could be fixed by expanding the tolerance zone but only as a last resort. The minimum tolerance is desirable for accuracy of reaching locations. 

This behavior can be observed by watching a rover repeatedly drive forward a short distance then turn 180 until it gets to the point.

wfvining, 2017-10-27:
This is because of the way we detect that the target was reached when in skid steer mode, which inherently requires overshooting the target. I have a potential fix where we also check the distance from that target implemented and tested, but need to do more testing to ensure it covers all corner cases and actually improves performance.

gmfricke, 2017-10-31:
Fixed in #114

closed 2017-10-31 by gmfricke