BCLab-UNM/SwarmBaseCode-ROS #87: Deployment script with -L applies changes on the second run

gmfricke, 2017-10-19:
@jschlind411, hoping you can look into this.

Run deploy.sh -L
Rover name/IP to Start: R05
Make change to source code on laptop. Recompile on laptop.
./run.sh on laptop
Rover name/IP to Start: R05
Changes don't seem to have been applied to the rover
Rover name/IP to Start: R05
Changes have been applied to the rover

JCHJones, 2017-10-24:
Could not reproduce when script is running as intended.
Could break if I assumed recompiling the code manually would move it to the rover.
However as the script is intended to be used the -RC command is required if a change was made.

This causes the script to repack the repository as just trying to start a rover in the order described above will use the old packaged tar.
I believe this is good behavior as repacking before every command to send to a rover is extraneous when starting multiple rovers.

Also the second time doesn't apply changes either I did it 10 times with no changes until I either used -RC or closed the script down and re-opened it causing it to do the first run compile and packaging.

gmfricke, 2017-10-24:
Thanks for checking this Jarett.

PR #107 (2017-12-12): Update README.md
Updated read me to better reflect script usage to avoid confusion on usage of -L command
fixes #87
closed 2017-10-24 by gmfricke