BCLab-UNM/SwarmBaseCode-ROS #131: Repeated simulations launched from same instance of GUI break

wfvining, 2017-11-09:
If you build a simulation, then click end simulation, and attempt to build a new simulation the rovers in the new simulation stay disconnected. They will not enter autonomous mode and will not accept manual commands; however, they appear (in the gui) to be receiving sensor data.

wfvining, 2018-11-20:
Might have been fixed. Need to test with latest code.

Antonio-Griego, 2018-12-21:
@wfvining 
Just for clarification, is this a bug for physical rovers, sim rovers, or both?
(I'm making the assumption its physical because that is your use case.)

wfvining, 2018-12-22:
@Antonio-Griego this is in sim. I'm not sure that this is still a problem, been a while since I ran repeated simulations.