BCLab-UNM/SwarmBaseCode-ROS #146: Step 5 Not Fully Going Through

asharma55, 2017-11-24:
When a tell catkin to build the Swarmathon Code, it does not work. It says that only 11 of the 13 packages were successfully installed. I am not sure of how I fix this.

wfvining, 2017-11-24:
Hello @asharma55, could you provide more information? 

* What packages failed
* What was the error

One common cause of errors is that the submodules have not been initialized or updated.

asharma55, 2017-11-24:
Here is what my terminal displays:
Errors     << rqt_rover_gui:make /home/asharma/SwarmBaseCode-ROS/logs/rqt_rover_gui/build.make.004.log
c++: internal compiler error: Killed (program cc1plus)

It later says two rover gui packages failed to install.

Failed     << rqt_rover_gui:make                 [ Exited with code 2 ]       
Failed    <<< rqt_rover_gui                      [ 1 minute and 0.5 seconds ] 
Abandoned <<< ublox_gps                          [ Unrelated job failed ]     
[build] Summary: 11 of 13 packages succeeded.                                 
[build]   Ignored:   1 packages were skipped or are blacklisted.              
[build]   Warnings:  None.                                                    
[build]   Abandoned: 1 packages were abandoned.                               
[build]   Failed:    1 packages failed.   

I am not sure of what the problem is. I have gone through all the other previous steps and everything worked fine but these two have failed for some reason.

mike-matera, 2017-11-25:
Hi,

Internal compiler error is often low memory or disk space related.

Cheers
./m

gmfricke, 2017-12-06:
I have seen this when I tried to compile under a virtual machine with only 2 GB of RAM allocated to the VM.

wfvining, 2017-12-11:
Please move these kinds of questions to the swarmathon forum.

Thanks, Mike, for replyinig.

closed 2017-12-11 by wfvining