BCLab-UNM/SwarmBaseCode-ROS #147: assembly ?

fonzg1, 2017-11-30:
does the competition allows use to use assembly language?

mike-matera, 2017-12-07:
Yes. Take a look at the competition rules here: 

https://github.com/BCLab-UNM/Swarmathon-Docs/blob/master/Competition%20Rules.md

Look for the section "Modifying the SwarmBaseCode-ROS code base". You can change anything you like within the allowable packages and I presume you can use any language you like. That said assembly code is a very poor choice for robot programming! Even for the Arduino.

wfvining, 2017-12-11:
Thanks for answering, Mike. 

@fonzg1 Please post these kinds of questions in the swarmathon forum in the future (http://nasaswarmathon.com/2018-qa-forum/?p=categories).

closed 2017-12-11 by wfvining

fonzg1, 2017-12-12:
> 2018-qa-forum/?p=categories).

Thank you Mike i appreciated, and yes Will Vining I'll do that from now on.