BCLab-UNM/SwarmBaseCode-ROS #151: Controller Address Changing

gmfricke, 2017-12-06:
The object address of the search controller changes over time (cout << this << endl). This means state in the controller is lost. There is no obvious recreation of the controller in LogicController.

JCHJones, 2017-12-06:
Is it possible that your actually getting the address of the pointer that points to searchcontroller

We regularly create new superclass controller pointers that point to each of subclass controllers as part of the priority reassignment. My best guess is that since this pointer is what the call to dowork is made using then that why the address is changing. That's my guess off the top of my head.

However if the cout was in say the set currentLocation function which calls searchcontroller by its actual reference the address wouldn't change.

JCHJones, 2017-12-07:
I attempted and was unable to replicate the changing adress

closed 2017-12-11 by wfvining