BCLab-UNM/SwarmBaseCode-ROS #172: Obstacle Avoidance Calls Not Being Updated

Antonio-Griego, 2018-02-12:
The obstacle avoidance calls number is not being properly updated in the GUI consistently. It does not update and remains at 0 while the rovers are running.

JCHJones, 2018-08-06:
@Antonio-Griego Lu Qi fixed this in CPFA and DDSA, can you update it in SwarmBaseCode

Antonio-Griego, 2018-12-21:
Tested with the Mars Diorama world file with simulated rovers.

The obstacle avoidance count does appear to update. However, I think the expected behavior of how often an update to the count occurs should be clarified.

For example, it _**seems**_ that each iteration the count is incremented while a rover is using the obstacle controller. And by iteration, I mean each cycle that the logic controller performs a single step and updates the controller.

Further testing with physical rovers may be necessary as well.