BCLab-UNM/SwarmBaseCode-ROS #187: Shutdown request received

fjrodl, 2018-07-04:
Hello, 

when running run.sh I always receive Shutdown request received and all the robots are disconnected:

`
[build] Summary: All 13 packages succeeded!                                    
[build]   Ignored:   1 packages were skipped or are blacklisted.               
[build]   Warnings:  None.                                                     
[build]   Abandoned: None.                                                     
[build]   Failed:    None.                                                     
[build] Runtime: 17.6 seconds total.                                           
robolab@robolab-XPS:~/SwarmBaseCode-ROS$ ./run.sh 
Running in /home/robolab/SwarmBaseCode-ROS
Cleaning up ROS and Gazebo Processes
Killing gzclient
Killing gzserver
Killing roslaunch
Killing rosmaster
Killing roscore
... logging to /home/robolab/.ros/log/cc2f04ae-7f65-11e8-b1cd-9cb6d0d18ad1/roslaunch-robolab-XPS-20212.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://robolab-XPS:33550/
ros_comm version 1.12.13

SUMMARY
========

PARAMETERS
 * /rosdistro: kinetic
 * /rosversion: 1.12.13

NODES

auto-starting new master
process[master]: started with pid [20223]
ROS_MASTER_URI=http://robolab-XPS:11311/

setting /run_id to cc2f04ae-7f65-11e8-b1cd-9cb6d0d18ad1
process[rosout-1]: started with pid [20236]
started core service [/rosout]
Rover GUI Starting...
[ INFO] [1530693605.343490740]: Finished loading Gazebo ROS API Plugin.
[ INFO] [1530693605.344153528]: waitForService: Service [/gazebo/set_physics_properties] has not been advertised, waiting...
Setting up world... done.
[ INFO] [1530693606.128709082, 0.177000000]: waitForService: Service [/gazebo/set_physics_properties] is now available.
[ INFO] [1530693606.243195419, 11.219000000]: Physics dynamic reconfigure ready.
[ INFO] [1530693613.926140896, 642.328000000]: Camera Plugin: Using the 'robotNamespace' param: '/'
[ INFO] [1530693613.933455492, 642.328000000]: Camera Plugin (ns = /)  <tf_prefix_>, set to ""
[ WARN] [1530693613.993402965, 642.328000000]: GazeboRosSkidSteerDrive Plugin (ns = //) missing <covariance_x>, defaults to 0.000100
[ WARN] [1530693613.993466617, 642.328000000]: GazeboRosSkidSteerDrive Plugin (ns = //) missing <covariance_y>, defaults to 0.000100
[ WARN] [1530693613.993496433, 642.328000000]: GazeboRosSkidSteerDrive Plugin (ns = //) missing <covariance_yaw>, defaults to 0.010000
[ INFO] [1530693613.993554391, 642.328000000]: Starting GazeboRosSkidSteerDrive Plugin (ns = //)
[ INFO] [1530693620.428855211, 659.273000000]: Camera Plugin: Using the 'robotNamespace' param: '/'
[ INFO] [1530693620.432115209, 659.273000000]: Camera Plugin (ns = /)  <tf_prefix_>, set to "achilles"
[ WARN] [1530693620.451071131, 659.273000000]: GazeboRosSkidSteerDrive Plugin (ns = //) missing <covariance_x>, defaults to 0.000100
[ WARN] [1530693620.451109950, 659.273000000]: GazeboRosSkidSteerDrive Plugin (ns = //) missing <covariance_y>, defaults to 0.000100
[ WARN] [1530693620.451128834, 659.273000000]: GazeboRosSkidSteerDrive Plugin (ns = //) missing <covariance_yaw>, defaults to 0.010000
[ INFO] [1530693620.451158886, 659.273000000]: Starting GazeboRosSkidSteerDrive Plugin (ns = //)
[ INFO] [1530693622.074153745, 660.602000000]: Camera Plugin: Using the 'robotNamespace' param: '/'
[ INFO] [1530693622.077282842, 660.602000000]: Camera Plugin (ns = /)  <tf_prefix_>, set to "aeneas"
[ WARN] [1530693622.196924198, 660.602000000]: GazeboRosSkidSteerDrive Plugin (ns = //) missing <covariance_x>, defaults to 0.000100
[ WARN] [1530693622.196972808, 660.602000000]: GazeboRosSkidSteerDrive Plugin (ns = //) missing <covariance_y>, defaults to 0.000100
[ WARN] [1530693622.196993365, 660.602000000]: GazeboRosSkidSteerDrive Plugin (ns = //) missing <covariance_yaw>, defaults to 0.010000
[ INFO] [1530693622.197021553, 660.602000000]: Starting GazeboRosSkidSteerDrive Plugin (ns = //)
[ WARN] [1530693623.638296458, 661.883000000]: Shutdown request received.
[ WARN] [1530693623.638416896, 661.883000000]: Reason given for shutdown: [new node registered with same name]
`

wfvining, 2018-07-05:
When you run the simulation you have robots that exist in the world, correct? Are you able to see all the topics associated with each robot when doing `rostopic list`? These topics should all be prefixed with the robot name (eg `/achilles/sonarLeft`).

fjrodl, 2018-07-06:
Hi 
Yes, I have the list of topics, but no data is published [Link](https://pastebin.com/gumnPkck)

This is the info that the rover interface shows: [Link](https://pastebin.com/LXwz6nkn)

The status of the rover interface is [Link](https://unsee.cc/ab7fd83f/) and Gazebo is showing the robots

gmfricke, 2018-07-09:
We were able to reproduce the error. It occurs after an apt upgrade of Ubuntu.

We will track down which apt package upgrade causes the bug.

wfvining, 2018-07-09:
As a temporary workaround if you run the simulation without visualization the robots appear to work normally.

fjrodl, 2018-07-09:
Which is the correct way to run the sim without visualization? Thanks!!

wfvining, 2018-07-10:
In the box where the build simulation/end simulation buttons are there is a check-box labeled "start visualization on build." If you un-check that you should be able to run the sim without visualization.

wfvining, 2018-07-10:
This appears to be caused by this issue:

[gazebo_ros_pkgs#765](https://github.com/ros-simulation/gazebo_ros_pkgs/issues/765)

Depending on the response to that issue, we could work around this by using `roslaunch gazebo_ros gazebo` when the viz is requested, and `roslaunch gazebo_ros gzserver` when no viz is requested. However, this would prevent us from starting the viz after the simulation is already going.

fjrodl, 2018-07-10:
Apparently, my run.sh dies in this scenario with the visualization on build "unchecked":

[ INFO] [1531241414.007267578, 775.179000000]: Camera Plugin (ns = /)  <tf_prefix_>, set to "aeneas"
[ WARN] [1531241414.123934040, 775.179000000]: GazeboRosSkidSteerDrive Plugin (ns = //) missing <covariance_x>, defaults to 0.000100
[ WARN] [1531241414.123977990, 775.179000000]: GazeboRosSkidSteerDrive Plugin (ns = //) missing <covariance_y>, defaults to 0.000100
[ WARN] [1531241414.124165937, 775.179000000]: GazeboRosSkidSteerDrive Plugin (ns = //) missing <covariance_yaw>, defaults to 0.010000
[ INFO] [1531241414.124206469, 775.179000000]: Starting GazeboRosSkidSteerDrive Plugin (ns = //)
terminate called after throwing an instance of 'std::out_of_range'
  what():  basic_string::erase: __pos (which is 18446744073709551615) > this->size() (which is 8)
./run.sh: line 22:  5720 Aborted                 (core dumped) rqt -s rqt_rover_gui
Cleaning up ROS and Gazebo Processes
killing:
 * /achilles_APRILTAG
 * /achilles_BASE2CAM
 * /achilles_BEHAVIOUR
 * /achilles_DIAGNOSTICS
 * /achilles_MAP
 * /achilles_NAVSAT
 * /achilles_ODOM
 * /achilles_SBRIDGE
 * /aeneas_APRILTAG
 * /aeneas_BASE2CAM
 * /aeneas_BEHAVIOUR
 * /aeneas_MAP
 * /aeneas_NAVSAT
 * /aeneas_ODOM
 * /ajax_APRILTAG
 * /ajax_BASE2CAM
 * /ajax_BEHAVIOUR
 * /ajax_MAP
 * /ajax_NAVSAT
 * /ajax_ODOM
 * /gazebo
 * /rosout
 * /rqt_gui_cpp_node_5720
[ WARN] [1531241420.967084982, 782.485000000]: Shutdown request received.
[ WARN] [1531241420.967194970, 782.485000000]: Reason given for shutdown: [user request]
[rosout-1] process has finished cleanly
log file: /home/robolab/.ros/log/36c2788c-8461-11e8-9d9c-9cb6d0d18ad1/rosout-1*.log
[rosout-1] restarting process
process[rosout-1]: started with pid [9189]

closed 2018-07-11 by gmfricke

ltenemeth, 2019-10-25:
> Apparently, my run.sh dies in this scenario with the visualization on build "unchecked":
>
> [ INFO] [1531241414.007267578, 775.179000000]: Camera Plugin (ns = /) <tf_prefix_>, set to "aeneas"
> [ WARN] [...]

Hi,
I'm getting the same error even after the fix provided for this issue. Was this the same for you as well? Thx.

PR #189 (2018-07-11): Assign the name gzclient to the gazebo_ros client node
Fixes #187 

Tested multiple start/stop viz scenarios. Ensured that simulated robots respond to auto/manual mode and that  they still collect cubes with/without viz.
commit 086803f8 (2018-07-11): Assign the name gzclient to the gazebo_ros client node

Fixes #187