BCLab-UNM/SwarmBaseCode-ROS #207: GUI crashes when click the odom in map frame for physical experiments

lukey11, 2018-09-24:
The GUI crashes when I clicked multiple options in the map frame.

wfvining, 2018-11-20:
@lukey11 can you give us a little more info? Also, is this still a problem?

Carter90, 2018-11-20:
We(Cabrillo) saw this problem, I have 2 suggestions `sudo` apt update; sudo apt upgrade` also I found that waiting until the rover was in a okay or idle state before clicking on it helped. 
If neither works I have found taking a look at the terminal where you launched the GUI and .ros/log folder in your home folder.

wfvining, 2018-11-27:
### To reproduce:
1. open pop-out map
2. select all four check boxes
3. click in pop-out map to bring to foreground
4. with pop-out in foreground click "manual" radio button

Antonio-Griego, 2018-11-27:
Possible fix from this commit from CPFA code:
https://github.com/BCLab-UNM/CPFA-ROS/commit/0e1a470dfc5fae570c275dfe80d5aab19417e7ac

Antonio-Griego, 2018-12-21:
Update 12/21/18 (testing with the current version of the base code)

Testing with virtual rovers doesn't appear to reproduce the bug reliably. However, it was observed that the camera feed stops updating (presumably crashed) when more than 1 rover is present during a simulation.

If anyone can add any additional details about their environments and settings for reproducing the bug I would certainly appreciate it!

TODO: Attempt to reproduce the bug with physical rovers, and try to narrow down the cause.

jalil29, 2019-02-16:
I see the same issue when I click the Odom check box during testing today.