BCLab-UNM/SwarmBaseCode-ROS #209: ublox broken

Carter90, 2018-10-15:
I wanted to test out your Rosbridge, as we have been running multimaster and I wanted to compare them. I did a fresh git clone and tried to build the basecode master. 
However I am unable to build ublox_msgs our rovers, not sure if its a ros-kinetic update or a kernel update. 

Reproduced with both our Hopper and Turing rovers running your SwarmathonIII image with an apt update, upgrade. Attached are the logs. 
[hopperUbloxCatkin.txt](https://github.com/BCLab-UNM/SwarmBaseCode-ROS/files/2480178/hopperUbloxCatkin.txt)
[turingUbloxCatkin.txt](https://github.com/BCLab-UNM/SwarmBaseCode-ROS/files/2480179/turingUbloxCatkin.txt)

gmfricke, 2018-10-18:
Hi Carter,

Which version of the ublox submodule is being pulled?

All the best,

Matthew

Carter90, 2018-10-18:
I cloned the UNM base code and it was at `2e16119` 
2e161196ad614a3b4f2f93caec3451d723af3613

Carter90, 2018-10-18:
If it is not causing problems for you I would suggest updating your ros packages.
Open `Software & Updates` then go to the `Other software` tab
make sure `http://packages.ros.org/ros/ubuntu` is checked
then run `sudo apt update;
sudo apt upgrade
`

gmfricke, 2018-10-22:
Hi Carter, 

I have been unable to reproduce the problem. Here is the log of a fresh clone and compilation just after running sudo apt update & sudo apt upgrade

The ublox hash is 2e161196ad614a3b4f2f93caec3451d723af3613 for ublox below.

Any ideas?

All the best,

Matthew

matthew@artemus:~$ git clone https://github.com/BClab-UNM/SwarmBaseCode-ROS
Cloning into 'SwarmBaseCode-ROS'...
remote: Enumerating objects: 99, done.
remote: Counting objects: 100% (99/99), done.
remote: Compressing objects: 100% (65/65), done.
remote: Total 15862 (delta 52), reused 59 (delta 33), pack-reused 15763
Receiving objects: 100% (15862/15862), 123.00 MiB | 16.92 MiB/s, done.
Resolving deltas: 100% (10443/10443), done.
Checking connectivity... done.
matthew@artemus:~$ cd SwarmBaseCode-ROS
matthew@artemus:~/SwarmBaseCode-ROS$ git submodule init
Submodule 'src/apriltags_ros' (https://github.com/BCLab-UNM/apriltags_ros.git) registered for path 'src/apriltags_ros'
Submodule 'src/ublox' (https://github.com/KumarRobotics/ublox.git) registered for path 'src/ublox'
Submodule 'src/usb_cam' (https://github.com/bosch-ros-pkg/usb_cam.git) registered for path 'src/usb_cam'
matthew@artemus:~/SwarmBaseCode-ROS$ git submodule update
Cloning into 'src/apriltags_ros'...
remote: Enumerating objects: 433, done.
remote: Total 433 (delta 0), reused 0 (delta 0), pack-reused 433
Receiving objects: 100% (433/433), 594.25 KiB | 0 bytes/s, done.
Resolving deltas: 100% (229/229), done.
Checking connectivity... done.
Submodule path 'src/apriltags_ros': checked out 'f51471b055b70012f86f9e73f67409d0a618c26a'
Cloning into 'src/ublox'...
remote: Enumerating objects: 1561, done.
remote: Total 1561 (delta 0), reused 0 (delta 0), pack-reused 1561
Receiving objects: 100% (1561/1561), 450.31 KiB | 0 bytes/s, done.
Resolving deltas: 100% (889/889), done.
Checking connectivity... done.
Submodule path 'src/ublox': checked out '2e161196ad614a3b4f2f93caec3451d723af3613'
Cloning into 'src/usb_cam'...
remote: Enumerating objects: 1139, done.
remote: Total 1139 (delta 0), reused 0 (delta 0), pack-reused 1139
Receiving objects: 100% (1139/1139), 508.49 KiB | 0 bytes/s, done.
Resolving deltas: 100% (461/461), done.
Checking connectivity... done.
Submodule path 'src/usb_cam': checked out '0c3c7cb65ce5bcbd8033f728b0d701825c87f9a0'
matthew@artemus:~/SwarmBaseCode-ROS$ catkin build
--------------------------------------------------------------------
Profile:                     default
Extending:             [env] /opt/ros/kinetic
Workspace:                   /home/matthew/SwarmBaseCode-ROS
--------------------------------------------------------------------
Source Space:       [exists] /home/matthew/SwarmBaseCode-ROS/src
Log Space:         [missing] /home/matthew/SwarmBaseCode-ROS/logs
Build Space:        [exists] /home/matthew/SwarmBaseCode-ROS/build
Devel Space:        [exists] /home/matthew/SwarmBaseCode-ROS/devel
Install Space:      [unused] /home/matthew/SwarmBaseCode-ROS/install
DESTDIR:            [unused] None
--------------------------------------------------------------------
Devel Space Layout:          linked
Install Space Layout:        None
--------------------------------------------------------------------
Additional CMake Args:       None
Additional Make Args:        None
Additional catkin Make Args: None
Internal Make Job Server:    True
Cache Job Environments:      False
--------------------------------------------------------------------
[... 49 lines cut ...]
  FindEigen3.cmake Module for the resulting CMake variable names.

Call Stack (most recent call first):
  CMakeLists.txt:21 (find_package)

cd /home/matthew/SwarmBaseCode-ROS/build/apriltags_ros; catkin build --get-env apriltags_ros | catkin env -si  /usr/bin/cmake /home/matthew/SwarmBaseCode-ROS/src/apriltags_ros/apriltags_ros --no-warn-unused-cli -DCATKIN_DEVEL_PREFIX=/home/matthew/SwarmBaseCode-ROS/devel/.private/apriltags_ros -DCMAKE_INSTALL_PREFIX=/home/matthew/SwarmBaseCode-ROS/install; cd -
...........................................................................................................................................................................
Finished  <<< usb_cam                              [ 8.6 seconds ]                                                                                                         
Finished  <<< apriltags_ros                        [ 18.2 seconds ]                                                                                                        
Starting  >>> behaviours                                                                                                                                                   
Finished  <<< ublox_msgs                           [ 18.6 seconds ]                                                                                                        
Starting  >>> rqt_rover_gui                                                                                                                                                
Starting  >>> ublox_gps                                                                                                                                                    
Finished  <<< behaviours                           [ 17.1 seconds ]                                                                                                        
[... cut ...]
[build]   Failed:    None.                                                                                                                                                 
[build] Runtime: 1 minute and 25.0 seconds total.                                                                                                                          
[build] Note: Workspace packages have changed, please re-source setup files to use them.
matthew@artemus:~/SwarmBaseCode-ROS$ ^C
matthew@artemus:~/SwarmBaseCode-ROS$

Carter90, 2018-10-23:
My only suggestion is to make sure you are getting update from http://packages.ros.org/ros/ubuntu

closed 2018-11-19 by wfvining