BCLab-UNM/SwarmBaseCode-ROS #210: Rosbridge corrections

Carter90, 2018-10-19:
executable bit missing on rover_launch_local.sh need `chmod +x ./misc/rover_launch_local.sh`
and incorrect relative path in rover_launch_local to rover_onboard_node_launch should be `./misc/rover_onboard_node_launch`

There were missing instructions on the readme possibly 
npm is not installed on Ubuntu defaultly need to run `sudo apt install npm`?

in swarmie-web.js
`ros : robotRos` change to `ros : swarmieName.robotRos,`
on lines 20, 26, 32, 38, 171, 240, 265, 290, 316
remove .toUpperCase() as most hostnames are lowercase 
on line 12

`dicument` should be `document` on line 377

wfvining, 2018-10-20:
the node.js stuff is cruft. I included it in the merge by mistake and none of it actually works yet. You shouldn't need npm to use this. 

`rover_launch_local` should be run from inside the misc dir. We should probably make it usable from anywhere, but that is another issue. If my memory serves me correctly `rover_on_board_node_launch` must be run from inside misc as well, but maybe that has been fixed.

wfvining, 2018-11-16:
Corrected in latest PR #216

closed 2018-11-16 by wfvining

Carter90, 2018-11-16:
Great, Cabrillo will take it for a spin tomorrow.

cabrillorobotics, 2018-11-17:
No luck on our part today getting it to run properly. Documentation was not clear.