BCLab-UNM/SwarmBaseCode-ROS #212: Make scripts more flexible

wfvining, 2018-10-20:
The launch and deploy scripts in the misc folder should be run-able from any directory. (spun off of issue #210)

Carter90, 2018-10-20:
The question is `any directory` do you mean from within the repo? Because what if you have multiple repos and does everyone have the same structure? 
I have various repos in a folder in my home directory. While the systems I image for the Cabrillo students are all in a folder called `rover_workspace` in their home directory.
You could add something to the sourced setup file to export an environmental variable and use that for the path of the directory. source `path to repo`/`repo`/`[devel|install]`/setup.[zsh|bash]
Keep in mind we have different directory structures, it may be in our home ~ or even on another drive, they may be building as devel or install and may even be using a different shell. 

Just for your own ease I would recommended just making everything work from the the repo base, this also makes it constant with your documentation.

Carter90, 2018-10-23:
Might also suggest using `$(catkin locate)`

Carter90, 2018-10-23:
@wfvining here is an example I made tonight where I used $(catkin locate), I had to change frameName for the sim and thought I might as well just regenerate all the Swamie models while I was at it I could fix the model names too. 

[buildSwarmieTemplates.txt](https://github.com/BCLab-UNM/SwarmBaseCode-ROS/files/2504778/buildSwarmieTemplates.txt)

wfvining, 2018-11-19:
Thanks Carter. I appreciate your suggestions. It might be simpler to just bring the scripts up to normal *NIX standards and avoid any catkin dependencies. If you have a pr with the suggestions above we would love to check it out, though. The initial issue is not well specified. All I want is to be able to type 
```
$ misc/rover_launch_local.sh
```
And have it work (this is basically what you mean by repo-root, I think).

Carter90, 2018-11-20:
I noted it because in the doc there are inconsistency sometimes the scripts would be run from the misc directory other times in the repo root. Right the root of the repo is the base folder where everything is contained. 
So my suggestion is whenever there is a relative path anywhere just use $(catkin locate)

Our misc is abit different. 
If I have time this week I will go through all the base code scripts and fix it.

Carter90, 2018-11-22:
Take a look at #221