BCLab-UNM/SwarmBaseCode-ROS #227: Simulated rover drives forward when turning in place at slow speeds

darrenchurchill, 2018-12-10:
Hey guys,

Since it's especially important this year to have the simulated and physical rovers behave similarly, I'd like to raise an issue we've seen with the simulated rovers during our tests.

Currently, if the simulated rover attempts to turn in place at ~0.8 rad/sec or slower, instead of simply turning in place, the rover creeps forward while turning. Sometimes the creep is large enough that the rover starts a 90 degree turn from a position completely outside of the home ring, and finishes it on top of the outer ring of April Tags.

Unfortunately, the Base Code's default turn speeds are fast enough to mask this discrepancy between the simulated and physical rover, both when teleoping through the GUI or letting the rover drive autonomously.

Turning in place using the keyboard teleop sends a Twist to the skidsteer plugin with angular z = +/- 4.5 rad/sec, while SBridge publishes Twists in autonomous mode that max out at ~ +/- 3.2 rad/sec and settle at ~ +/- 1 rad/sec when the rover is performing an in-place turn. So, I don't expect you to have observed this in your tests, but you can still replicate the issue without modifying any source code by publishing to the skidsteer topic directly:

```
rostopic pub /achilles/skidsteer geometry_msgs/Twist "linear:
  x: 0.0
  y: 0.0
  z: 0.0
angular:
  x: 0.0
  y: 0.0
  z: 0.8"
```

You could also use `rosrun rqt_robot_steering rqt_robot_steering`. The interface lets you type the topic to publish on, set min and max values for the Twist, and has convenient keybindings (hover over the buttons for the tooltips). Set the max angular vel to +-0.8 rad/sec and turn the rover back and forth from it's starting position. Eventually it will have moved all the way onto the home ring. 

The creeping gets worse with even smaller values. It's also present regardless of how fast the simulation is running.

I hoped that simply forcing the simulation to run at real time, while possibly increasing the number of ODE solver iterations from 50 to 100+ would fix the issue (`gz physics -i 100` at runtime). Unfortunately, 100 iterations doesn't really remove the creep, especially at speeds slower than 0.8 rad/sec, and pushing the setting much higher with a step size of 0.001 makes it pretty hard to run at a 1x real time factor while simulating multiple rovers.

I did, however, find two ODE parameters that fix this problem. The Constraint Force Mixing parameter (CFM) and Error Reduction parameter (ERP) are sometimes used to model spongy or springy joints, but they can also be used together to reduce numerical errors in the simulation. See http://www.ode.org/joints.pdf

A CFM of 0 (the global default), puts a hard constraint on a joint, while a positive value allows the joint to be pushed on to violate its now soft contraint. You can visualize this parameter in action if you open Gazebo Client and set the global CFM in physics > constraints (try 0.2 for example, it's kind of fun! Probably cap the update rate at 1000 to slow everything down). The whole rover moves as though all the joints are connected by springs, and you can see the wheels pull away from the base during a turn.

The ERP determines how quickly any joint error will be reduced. An ERP of 1 will, in theory, eliminate any joint error in one time step. The global default ERP is 0.2, and recommended values are 0.1 to 0.8 according to the linked documentation. Again, to see what the ERP does to the rover's model, try setting it to an entertaining 0.05, while leaving the CFM at 0.2. Now the wheels can pull completely away from the base, and almost pull it along the ground, but you can see they do still return slowly to their proper position.

I think using a CFM of 0.01 and an ERP of 0.8 dramatically improves the rover's ability to turn in place without driving forward. The CFM is small enough so the rover doesn't look like it has a full suspension, and the ERP is large enough to quickly bring the softened joints back to their proper position. Additionally, you can set these parameters on individual joints. I set them on each of the four wheel joints, while leaving the default global values, and the turning performance is equivalent, but setting the parameters on just the wheel joints virtually eliminates any of the residual bounciness that using the global parameters can result in.

Using a larger ERP for the wheel joints also gives you the added benefit of preventing the errors from intermittently growing large enough so the rover moves forward on its own. Sometimes I'll forget the simulation is running, and come back to see that the rover has somehow driven forward into the home ring when it should have been stopped.

I built the base code and recorded a video of two rovers turning back and forth at 0.8 rad/sec in the simulation. Achilles (black) has the current model, and Aeneas (yellow) has the CFM and ERP parameters set for each wheel joint.

https://youtu.be/_CpjMkTUQls

You can see in the video, Achilles moves all the way onto the home ring, while Aeneas doesn't move forward at all.

I pushed a branch to my fork with the update to Aeneas if you want to check it out and start the simulation with one original model and one updated model for comparison.

https://github.com/darrenchurchill/SwarmBaseCode-ROS/tree/sim-cfm-erp-wheels

Is this a change you guys would consider?

wfvining, 2018-12-19:
@darrenchurchill Thank you for your awesome effort to make the simulation more realistic! I think this is a change that we would consider; however, I need @gmfricke and others to weigh in on this before we move ahead.

Sorry for the slow response, it has been a busy time and there aren't as many of us working on the swarmathon as there used to be.

darrenchurchill, 2018-12-21:
No problem, Will, I understand. Thanks for getting back to me. Let me know what you guys think, and whether you'd like a full pull request with all the models updated.

gmfricke, 2019-01-08:
Let's discuss this at our next Swarmathon dev meeting.

Antonio-Griego, 2019-01-14:
@darrenchurchill This is awesome! We would love to see this fix implemented. Could you do a pull request with this update? We would love to test it and approve it.