BCLab-UNM/SwarmBaseCode-ROS #233: Node name collisions in swarmie.launch

darrenchurchill, 2019-01-04:
Hey guys,

I saw that you merged a lot of pull requests this afternoon, and it looks like 3dc05c7429e5a3ca9eca2d5a08c0de286f74d139 introduced an error. The topic_tools throttle nodes all launch in the global ROS namespace with the same name, so every time a new rover comes up in the simulation, the previous rover's throttle nodes receive shutdown requests due to name collisions.

This causes the GUI to only be able to subscribe to sensor data from the rover brought up most recently. Prefixing node names with 'rovername_' like is done with the rest of the nodes should fix the problem.

wfvining, 2019-01-05:
Thanks for catching this @darrenchurchill. Should be a simple fix. Feel free to submit a PR if you have time, otherwise I will get to it Monday or Tuesday.

gmfricke, 2019-01-08:
Thanks for catching this! Resolved by #235.

closed 2019-01-08 by gmfricke