BCLab-UNM/SwarmBaseCode-ROS #236: Publishing String on mode topic in run_headless_sim.sh

darrenchurchill, 2019-01-06:
https://github.com/BCLab-UNM/SwarmBaseCode-ROS/blob/55d16639e6c1a863d3dc15054c5cac1e95c093b6/run_headless_sim.sh#L337

We ran into an issue with this message type conflict in using a modified version of this script. Messages on the `mode` topic should be a `std_msgs/UInt8`

Carter90, 2019-01-06:
Produces the errors 

`
[rospy.internal][ERROR] 2019-01-04 23:16:11,363: Unable to initiate TCP/IP socket to RobotKey:46489 (http://RobotKey:41107/): Traceback (most recent call last):
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/rospy/impl/tcpros_base.py", line 557, in connect
    self.read_header()
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/rospy/impl/tcpros_base.py", line 650, in read_header
    self._validate_header(read_ros_handshake_header(sock, self.read_buff, self.protocol.buff_size))
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/rospy/impl/tcpros_base.py", line 593, in _validate_header
    raise TransportInitError("remote error reported: %s"%header['error'])
TransportInitError: remote error reported: topic types do not match: [std_msgs/UInt8] vs. [std_msgs/String]`

`[WARN] [1546672475.265579]: Could not process inbound connection: topic types do not match: [std_msgs/UInt8] vs. [std_msgs/String]{'message_definition': 'uint8 data\n', 'callerid': '/achilles/mobility', 'tcp_nodelay': '0', 'md5sum': '7c8164229e7d2c17eb95e9231617fdee', 'topic': '/achilles/mode', 'type': 'std_msgs/UInt8'}
[WARN] [1546672475.321932]: Could not process inbound connection: topic types do not match: [std_msgs/UInt8] vs. [std_msgs/String]{'message_definition': 'uint8 data\n', 'callerid': '/aeneas/mobility', 'tcp_nodelay': '0', 'md5sum': '7c8164229e7d2c17eb95e9231617fdee', 'topic': '/aeneas/mode', 'type': 'std_msgs/UInt8'}
[WARN] [1546672475.367564]: Could not process inbound connection: topic types do not match: [std_msgs/UInt8] vs. [std_msgs/String]{'message_definition': 'uint8 data\n', 'callerid': '/ajax/mobility', 'tcp_nodelay': '0', 'md5sum': '7c8164229e7d2c17eb95e9231617fdee', 'topic': '/ajax/mode', 'type': 'std_msgs/UInt8'}
[WARN] [1546672475.409271]: Could not process inbound connection: topic types do not match: [std_msgs/UInt8] vs. [std_msgs/String]{'message_definition': 'uint8 data\n', 'callerid': '/diomedes/mobility', 'tcp_nodelay': '0', 'md5sum': '7c8164229e7d2c17eb95e9231617fdee', 'topic': '/diomedes/mode', 'type': 'std_msgs/UInt8'}`

gmfricke, 2019-01-08:
Thanks for catching this. Resolved by #237.

closed 2019-01-08 by gmfricke