BCLab-UNM/SwarmBaseCode-ROS #245: Output of SearchController does not match the behavior of the Swarmie

jalil29, 2019-01-26:
I recently migrated to the new 1.4 Swarmie code after using the previous version to setup the search controller, but some change resulted drastically different behavior from the clean state it was in before.

For example,  (ODOM map) if I move from (0,0) to (1,1) to (-3,5) there would be a straight line between each point.
Now those lines are not straight, in fact it backtracks and zigzags( for lack of a better description ).

before 1.4
[image]

after update
[image]

I could provide the full image upon request, but preferably not in a public forum

I've spent time looking at the issue, but the only reasoning I can come up with is the change from using twist to the new skid message.

gmfricke, 2019-01-26:
> capture
> capture 1

Thanks for reporting this. Are you testing in simulation or on physical
rovers?

jalil29, 2019-01-26:
Simulation

jalil29, 2019-01-28:
Think i discovered the issue, it seems to be related to a value that was inherited from the 2018 code we used that I have not touched. Noticed the number as was making changes line by line to discover the issue.

jalil29, 2019-01-29:
Would anyone know why result.wpts.waypoints[0].x or result.wpts.waypoints[0].y would deviate from the input that was put in by the search controller? 
the input and current location mostly remains within the hypo()<.15 error, but with what happens elsewhere in the code would cause the error to exceed the .15 margin.

gmfricke, 2019-02-18:
@JCHJones any ideas what might be rewriting the values?