BCLab-UNM/SwarmBaseCode-ROS #253: rosout-1 error, continous restarting process

jalil29, 2019-03-23:
Talked about this during the tech session today as an occasional bug that seems to occur.

[image]

[093c4a7e-4d0a-11e9-8dd4-fcaa147aacd3.zip](https://github.com/BCLab-UNM/SwarmBaseCode-ROS/files/2998886/093c4a7e-4d0a-11e9-8dd4-fcaa147aacd3.zip)

Carter90, 2019-04-09:
The logs indicate you are trying to start a node that is already running with the same name. 
I don't see any rovers simulated or physical in the logs. /use_sim_time is not set so that means you intend on deploying to physical swarmies. Are they already deployed are you trying to use ros bridge? 
Are you starting roscore separately? 
What is your procedure to reproduce this issue?

wfvining, 2019-04-10:
I haven't had a chance to look at your logs, but I recently saw a similar problem with restarting nodes. In that case the problem was that I was publishing the wrong message type on a topic, might be worth real quick.

jalil29, 2019-04-10:
It happens when I open the sim software (./run.sh) before i can simulate the robots, doesn't happen every time

Carter90, 2019-04-10:
Are you running the stock(SwarmBaseCode-ROS) `run.sh` script or have you made modifications?
Are you using the stock GUI and have ran `catkin clean -y` and then `catkin build` since the GUI update?
Quite awhile ago the old basecode's the cleanup script was missing the execution bit and when you called ./cleanup.sh (run.sh does twice) it would cause problems. When you run `./cleanup.sh` do you get`./cleanup.sh: Permission denied`? if so run `chmod +x ./cleanup.sh`

jalil29, 2019-04-10:
practically stock, only added a ". devel/setup.bash" line to it at the beginning even though it's useless since i can't listen to rostopic from that terminal during the simulation anyway.
I have ran "catkin clean -b" in the past, not sure if I've done so since gui update/ last time i reinstalled ubuntu 
`jalil@MainPC:~/master2019$ ./cleanup.sh 
Killing gzclient
Killing gzserver
Killing roslaunch
`
cleanup seems fine

Carter90, 2019-04-10:
do you mean `./devel/setup.bash`?
If you want to actually do something you should have `source ./devel/setup.bash` instead.
Do a `catkin clean -y` and then `catkin build` and let me know if you still have the problem opening the GUI

jalil29, 2019-04-10:
GUI does open it just keeps doing that despite the GUI being open and simulating properly i believe

Carter90, 2019-04-10:
And the rover is not shuddering? When you are running the sim what is the output of `rosnode list`

wfvining, 2019-04-10:
@Carter90 `. foo` is an alias for `source foo` in bash.